mister-grumbler/w1209-firmware #2: Build string to display temperature

mister-grumbler, 2017-11-15:
A function which would convert temperature value (positive/negative, with/without tenth part of degrees) into valid string.

pradoluis, 2017-11-20:
int getTemperature() {
  average = 1023 / average - 1;
  average = 20000 / average;
  float steinhart;
  steinhart = average / 10000;     // (R/Ro)
  steinhart = log(steinhart);      // ln(R/Ro)
  steinhart /= 3950;               // 1/B * ln(R/Ro)
  steinhart += 1.0 / (298.15); // + (1/To)
  steinhart = 1.0 / steinhart;     // Invert
  steinhart -= 273.15;             // convert to C degrees Celsius
  steinhart *=10;
  return  = steinhart;
}

mister-grumbler, 2017-11-21:
Own version of itoa () was implemented. Not because author think himself a very smart. But because of project requirements.

closed 2017-11-21 by mister-grumbler