mister-grumbler/w1209-firmware #19: New board has common anode LED display

zcsahok, 2018-02-04:
I have recently purchased a board on Ali and uploading your firmware produced a randomly garbled display.

I turns out that my board has a common anode (CA) LED display (p.n. **TUSO-02803-U**) as opposed to the common cathode (CC) display on the earlier boards.

The assignment of following pins was changed

Pin | Original (CC) | New (CA)
----|------------------|------------
SEG_B  | PA.2 | PB.5
SEG_F | PA.1 | PB.4
DIGIT_1 | PB.4 | PA.2
DIGIT_2 | PB.5 | PA.1

For the CA display the segments are active low and the digits are active high.

With the above changes firmware works as expected, thanks for sharing it.

mister-grumbler, 2019-07-04:
Thank you for the report. And sorry for such a big delay.
Could you please attach pictures (both sides) of your board? I wish to put them on WiKi with appropriate comments.
If it possible, put the board over white paper when taking a picture.
Thank you in advance for that!

Regarding code modifications: I will make appropriate changes and place them within _#ifdef_ macros. So it will be possible to choose the type of LED display during compilation.

zcsahok, 2019-08-10:
Please check this wiki summarizing the variants:
https://github.com/TG9541/stm8ef/wiki/W1209-Identifying-'compliant'-boards

mister-grumbler, 2019-08-13:
Thank you very much for your participation!