mister-grumbler/w1209-firmware #20: Is this a STVP project? or IAR?

xlichen, 2020-02-20:
Thanks!

TG9541, 2020-02-20:
none of these: 
`CC       := /usr/bin/sdcc`

xlichen, 2020-02-21:
You've used "Small Device C Compiler" to generate hex file, then use Burn Tool to put the file into STM8?

TG9541, 2020-02-21:
That's certainly a way to do it :-)
This here might help: https://github.com/TG9541/stm8ef/wiki/STM8-eForth-Programming

santosh1502, 2021-03-12:
is it possible for some one to provide hex also i have compiled in stvd myP-1 function is not working

Diegocn9, 2022-04-22:
Tem como mudar a configuração dessas plaquinha para ser usada como temporizador acumulativo por pulso?

rtek1000, 2023-03-24:
> Tem como mudar a configuração dessas plaquinha para ser usada como temporizador acumulativo por pulso?

Oi, sim, tem como reprogramar.

Hi, yes, there is a way to reprogram it.